OPEnSLab-OSU/Loom #13: Loom Error Handeling

Cogswatch, 2019-10-09:
Add SD Logging of Error handling or edge case events, consider implementing watchdog longjump for failure recovery.

prototypicalpro, 2019-10-09:
Loom doesn't currently keep a ledger of previous events (sensor creation, errors, etc.) and as a result it can be difficult to troubleshoot what is going wrong. Loom should be able to store a history of events on another medium other than the micro controller, such that a developer can access these logs after a failure.

Events should include:
 * Sensor creation/destruction
 * Timeouts
 * Errors
 * Metadata/information
 * Hardware status (free memory)
 * An accurate timestamp, or at least a useful relative one.

Cogswatch, 2020-04-01:
This can be solved with FeatherFault, consider implementing it into core Loom to resolve this issue.

udellc, 2022-01-20:
FeatherFault is integrated in Loom3, addresses many of these features

closed 2022-01-20 by udellc