OPEnSLab-OSU/Loom #17: Errors from not having all libraries downloaded

veacha, 2019-10-10:
I tried to compile an example and had errors from not having each library downloaded in the IDE on PC. Opened library manager and downloaded each one as they came up. Could there be a list of all libraries needed in order to work with Loom with set-up instructions?

Cogswatch, 2019-10-11:
Try heading over to the [**Install the Loom Dependencies**](https://github.com/OPEnSLab-OSU/Loom/wiki/Loom-Installation#install-the-loom-dependencies) section on our wiki, which provides details on how to acquire all the library dependencies. We are currently tailoring development towards automating the process rather than expecting users to perform the install manually. Does this help?

veacha, 2019-10-22:
Yes this helps. Thank you

closed 2019-10-22 by prototypicalpro