OPEnSLab-OSU/Loom #18: Loom Analog Bug: InvalidInput

osuwalkera6, 2019-10-11:
When I upload my code using the Arduino IDE, it compiles and load onto the board just fine. However, looking at the serial monitor for readings, I receive the error: 

"[Device] deserializeJson() failed: InvalidInput"

I notice that the Json configuration for Analog "General" is different for past projects than it is in the Loom Master Directory now. I tried changing the inputs to specific values for the array to no avail. Hopefully you guys can figure this out!
[Analog_Bug_code.docx](https://github.com/OPEnSLab-OSU/Loom/files/3719751/Analog_Bug_code.docx)

goertzel, 2019-10-11:
Deserialization errors result from invalid Json configurations (and currently cannot be caught until the device tries to parse it). In your case the issue seems to be the use of 'Conversion::NONE'. Unfortunately you cannot use enum values directly in the config, but have to use the int representation instead, as you did with A2 (which your are presumably using with a turbidity sensor). So you should be able to just switch 'Conversion::NONE' to 0 to fix the issue.

As far as the general configuration settings go, what are you noticing as different? There were some unnecessary device id info that was removed from the config (but it won't hurt if you still have that information in the config) and interval was added. Interval goes with Loom.pause() as the number of milliseconds to wait between loop iterations.

prototypicalpro, 2019-10-16:
Closing this issue, feel free to open it again if you are still having problems.

closed 2019-10-16 by prototypicalpro