OPEnSLab-OSU/Loom #21: Adafruit I2C Soil Moisture Sensor Integration Request

maxchu765, 2019-10-15:
I would like these new I2C soil moisture sensors to be integrated. 

https://www.adafruit.com/product/4026

maxchu765, 2019-10-16:
I tested the sensor with Adafruit's example code and it works, There is a library for the sensor. Here is the link with the example code.
https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/arduino-test

prototypicalpro, 2019-10-22:
Additionally add support to for the I2C multiplexer plugging.

eliwinkelman, 2019-10-31:
@maxchu765 Is there one of these sensors I can test the integration with?

maxchu765, 2019-10-31:
@eliwinkelman There's a bunch in the RFID moisture project cabinet that you can grab.

eliwinkelman, 2019-11-01:
This is now in the development branch of Loom

closed 2019-11-01 by eliwinkelman