OPEnSLab-OSU/Loom #28: AGU Loom Kits needed by December 2

udellc, 2019-10-21:
**Project Name**
AGU Loom Kits
**Project Description**
We need 20 Loom kits for the AGU conference.

6 x hydra kits w/ PCB
- Pressure+Temp
- pH
- Turbidity
- EC

6 x terra kits w/ PCB
- Cap Soil moisture (i2c)
- Meter GS 3

6 x atmos kits w/ PCB
- Temp-humidity (SHT31D)
- Solar Radiation (TSL2591)

Each kit will have common: 
**Hardware List**
* Feather M0 WiFi
* Hypnos Board
* Coincell battery
* OLED Shield
* SD card
* 2000 mAh battery
(Maybe)
* Pelican case
* Panel mount Jacks

**Additional Project Requirements**
Base functionality:
Plug Sensors into Box
Log to SD
Log to Google Docs via WiFi
Report Data over LAN to MaxMSP for visualization

* Things the project must be able to do to succeed
To be discussed, added in meeting.

* The more detail, the more we will be able to aid in development

prototypicalpro, 2019-11-21:
An updated project specification is as follows:

## Overview

All kits must have the following functionality (in order of priority):
 * Read accurate data from all sensors available (including time)
 * Display this data to the OLED screen in an understandable manner
 * Given a wireless connection, publish this data to it's individual tab in a single Google Sheet
 * Given a wireless connection, publish this data to the Spool platform
 * Continue reading accurate data despite sensors being removed or missing during operation.
This functionality must be present for any hardware configuration allowed in the specification below.

## Hydro Kit Component List

* Feather M0 WiFi (any WiFi firmware)
* Hypons Board w/ coin cell (DS3231, I2C standard pins, `0x68`) (SDCard, SPI standard pins + CS on 10)
* OLED (FeatherWing, I2C standard pins, `0x3C`)
* Pressure+Temp (MS5803, I2C Standard pins, `0x76`)
* pH (DF Robot Ph v1.1, Analog Pin A1)
* Turbidity (DFRobot Turbidity, Analog Pin A2)

## I2C Kit Component List

* Feather M0 WiFi (any WiFi firmware)
* Hypons Board w/ coin cell (DS3231, I2C standard pins, `0x68`) (SDCard, SPI standard pins + CS on 10)
* OLED (FeatherWing, I2C standard pins, `0x3C`)
* I2C Multiplexer (TCA9548Ab, I2C Standard Pins, `0x71`)
* Temp-humidity (SHT31D, I2C Through Multiplexer, `0x44`, Any Port)
* Solar Radiation (TSL2591, I2C Through Multiplexer, `0x39`, Any Port)
* Soil Moisture (Adafruit STEMMA, I2C Through Multiplexer, `0x36`, Any Port)
* Ultrasonic sensor (MB1232, I2C Through Multiplexer, `0x70`, Any Port)

prototypicalpro, 2019-12-21:
Completed, feedback to be posted as individual issues.

closed 2019-12-21 by prototypicalpro