OPEnSLab-OSU/Loom #29: Provide support for parsing TOML (Tom's Obvious, Minimal Language) formated config

Cogswatch, 2019-10-23:
**Is your feature request related to a problem? Please describe.**
JSON, while incredibly useful from a developer position and fundamentally simple, is a relatively hostile standard for someone who has no development experience. TOML is a standard that works 1:1 with JSON, while appearing far more readable and friendly (for example, inline comments!).

**Describe the solution you'd like**
Identify a **lightweight** toml to json parser to sit on top of ArduinoJSON in order to provide TOML config input and output

**Describe alternatives you've considered**
Arduino JSON by default provides support for NLDJSON (new line delimited JSON) as well as a similar, near bracket less JSON format, despite how much these provide useful features, they are not quite the easily approachable format that TOML can provide.

**Additional context**
https://github.com/toml-lang/toml#user-content-example

prototypicalpro, 2019-10-23:
How would TOML translate into a header file? One reason JSON was originally chosen is because it is not newline delimited by default, and as a result can be stored in structured format in a header file.

prototypicalpro, 2019-10-23:
TOML does not translate well to a header file, and as a result this can be tabled until the configurator web interface is built.

closed 2019-10-23 by prototypicalpro