OPEnSLab-OSU/Loom #31: Add API for more flexible data manipulation outside of Loom

prototypicalpro, 2019-10-23:
**Is your feature request related to a problem? Please describe.**
Multiple people have approached the Loom development team about accessing the outputs of the sensors directly. While Loom is planned to include most of the functionality a developer would need through the JSON interface, some underlying functionality should be exposed in the sketch while the action system is still in development.

**Describe the solution you'd like**
A `get_json` function or similar that retrieves the JSON generated by measuring sensors. This function would return a _copy_ of the JSON, so that if any modification is made it does not affect the stability of Loom. Additionally, `publish`, `display_data`, and any other functions using the internal Loom JSON should have a variant that operates on an external JSON object. These functions will also require documentation on what JSON format is required for them to execute successfully.

**Describe alternatives you've considered**
This solution is an alternative to the JSON metamodules/actions system, which is currently in development and will be for some time.

Cogswatch, 2019-10-23:
The LoomManager::Packager() function returns the packaged JSON, which then can be modified as the user pleases, after which, the LoomManager::Publish(JsonDocument) and/or LoomManager::Log(JsonDocument) functions can be used to return the data back into the loom ecosystem. For now, users should refer to the [LoomManager Documentation](https://openslab-osu.github.io/Loom/html/class_loom_manager.html) for more detail on how to access and modify Loom data.

That said, a formal wiki page dedicated to the subject is probably long overdue, re-tagging issue appropriately.

prototypicalpro, 2019-11-13:
This seems to have been addressed, I'm closing this issue.

closed 2019-11-13 by prototypicalpro