OPEnSLab-OSU/Loom #32: Accessing the JSON object returned from LoomManager::package hangs device

prototypicalpro, 2019-10-24:
**Describe the bug**
`LoomManager::package()` should return a JSON object reference, so the user can access the JSON created by the Loom sensors. When accessing an element of this JSON reference, the device hangs, suggesting undefined behavior.

**Hardware in Use**
Any feather m0.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the Loom "Basic" example sketch.
2. Replace `Loom.package()` with the following:
```C++
auto thing = Loom.package();
auto sensor = thing["contents"][1]["data"];
LPrintln(sensor.isNull());
```
3. Device will hang after "Setup Complete"

**Expected behavior**
The `sensor` variable should contain data from the Analog sensors as a JSON object.

prototypicalpro, 2019-10-24:
Replacing the above code with `(thing["contents"].as<JsonArray>())[1]["data"]` fixes the error. I'm not sure if this is a problem with Loom or ArduinoJson.

Cogswatch, 2019-10-25:
This is less a problem with ArduinoJson as it is a syntax error. The way ArduinoJson was designed, you have to do it that way. I'm not sure if JsonDocuments support subscript integer access, you might need to hand it a string'd int in order to do that, but besides that, only JsonArrays, which you cast to, support that syntax. Beyond that, nothing here is working incorrectly, the correct way just isn't ideal.

prototypicalpro, 2019-10-25:
It is not a syntax error, see arduinojson documentation: https://arduinojson.org/v6/api/jsondocument/subscript/ https://arduinojson.org/v6/api/jsonvariant/subscript/ . It might be a bug with ArduioJson which is causing undefined behavior, but according to the documentation above the syntax suggested should work.

prototypicalpro, 2019-10-28:
The problem is caused by using `auto` type when accessing any element of the Json. I've submitted an issue to ArduinoJson (https://github.com/bblanchon/ArduinoJson/issues/1120), however in the meantime we will need to be careful with our use of `auto`.

prototypicalpro, 2019-10-30:
Issue has been fixed, and will be in the next ArduinoJson relase.

prototypicalpro, 2019-11-01:
Fixed in [ArduinoJson v6.13.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.13.0).

closed 2019-11-01 by prototypicalpro