OPEnSLab-OSU/Loom #36: Add FAQ to the Loom homepage

prototypicalpro, 2019-10-30:
**Is your feature request related to a problem? Please describe.**
A number of people have approached the Loom team stating that they've had trouble finding the answer to their question in the documentation. Most of the complaints have centered around the structure of the documentation being confusing or difficult to navigate.

**Describe the solution you'd like**
It was suggested that a FAQ be created as a starting page for questions about Loom. This page would cover all questions we have received about Loom, provide a detailed answer to each question, and link to the appropriate documentation for a more detailed explanation. Some topics that have already been suggested for an FAQ:
 * How do I add my sensor to Loom?
 * I found a bug! What do I do?
 * Why is Loom so slow to compile? Is there any way to make it faster?
 * I want to get/add data for X sensor for Loom. How do I do that?
 * I keep getting duplicate library errors. Help!

eliwinkelman, 2019-11-01:
Adding to this that we should probably split the FAQ into two parts: common questions and troubleshooting.

Things like "How do I add my sensor to Loom?" go in common questions and troubleshooting would be things like "Loom says my configuration isn't valid (common error message)".
Ideally troubleshooting would include a short paragraph for every common error message in Loom explaining why that error usually occurs and how it can be fixed.

eliwinkelman, 2019-11-01:
There is a proposed list of starting questions for the FAQ on this page: https://github.com/OPEnSLab-OSU/Loom/wiki/Frequently-Asked-Questions

closed 2020-07-03 by MrKangs