OPEnSLab-OSU/Loom #37: SDI-12 Soil Moisture Sensor

maxchu765, 2019-10-30:
**Item's use**
SDI-12 is the communication protocol that is used by the Teros and Decagon sensors. 

**Product Page**
[Teros 11/12 Integrator Guide](http://manuals.decagon.com/Integrator%20Guide/18224%20TEROS%2011-12%20Integrator%20Guide.pdf)

[SDI-12 Spec](http://www.sdi-12.org/archives/SDI-12_version1_3%20January%2028,%202016.pdf}
**Existing Implementation**
[Library used in LOOM 1) (https://github.com/EnviroDIY/Arduino-SDI-12)

prototypicalpro, 2020-01-04:
Where are you at with this issue? Do you need help with this integration?

maxchu765, 2020-01-06:
I have the sensors dirty integrated, but will need help to get them properly integrated into LOOM.

maxchu765, 2020-01-09:
https://github.com/OPEnSLab-OSU/SitkaNet-2

closed 2022-01-25 by WL-Richards