OPEnSLab-OSU/Loom #42: LoRa m0 and Ethernet featherwing conflict

eliwinkelman, 2019-11-07:
**Describe the bug**
When using Loom you can't use the LoRa m0 and ethernet featherwing at the same time. Ethernet initialization fails whether or not LoRa capabilities are enabled.

**Hardware in Use**
Adafruit Feather M0
Ethernet FeatherWing

**To Reproduce**
Steps to reproduce the behavior:
1. Stack the ethernet featherwing on the feather m0.
2. Plug the ethernet into an ethernet connection.
3. Run a simple loom config with only ethernet enabled (make sure you have a proper mac address, etc.).
4. See the error in the serial with ethernet initialization failing.

**Expected behavior**
We expect ethernet to initialize and work fine. There aren't any pin conflicts that I could find with LoRa.

**Code**
Google Sheets Ethernet example demonstrates the problem.

**Config**
Google Sheets Ethernet example demonstrates the problem.

**Additional context**
We know that this is a problem with Loom because the old eGreenhouse code runs ethernet and LoRa (both enabled) at the same time just fine.
The only difference I could find in the LoRa code is that loom sets the LoRa CS Pin (pin 8) to pullup high, but as far as I can tell, ethernet doesn't use this pin.

prototypicalpro, 2020-01-10:
I think this is a symptom of the Radiohead library not deactivating the CS pin for the LoRa module properly. I was able to address the problem by deselecting the LoRa module before any interaction with the Ethernet module, however this should only serve as a band-aid for now.

It should also be noted that you _cannot_ use Ethernet at the same time that you are expecting to receive a packet over LoRa with this fix. This is because LoRa can trigger an interrupt and take the SPI bus anytime it detects a packet, which may or may not be during an SPI transaction with the Ethernet module. In the future, this problem needs to be addressed by either splitting the SPI buses, or creating a mutex system that both ethernet and lora can use. 

Code is here: https://github.com/OPEnSLab-OSU/Loom/pull/59

prototypicalpro, 2020-01-17:
Fixed the above problem by sleeping the LoRa modem when not transmitting/receiving. Code is here: https://github.com/OPEnSLab-OSU/Loom/pull/62

closed 2020-01-17 by prototypicalpro