OPEnSLab-OSU/Loom #44: DS3231 RTC reports wrong time

maxchu765, 2019-11-08:
**Describe the bug**
The DS3231 RTC reports the wrong time. The offset varies from as little as 20 min up to over 9 hours with different Hypnos boards. I tested 4 different Hypnos boards, all showed the same issue with different offsets. However, all 4 RTCs worked corrently using the RTClib DS3231 example code.

**Hardware in Use**
Feather M0 Proto
Hypnos

**To Reproduce**
Steps to reproduce the behavior:
1. Stack Hypnos and Feather M0 Proto
2. Upload Hypnos_SD example
3. See error

**Expected behavior**
The RTC should report the correct time.

**Code**
Hypnos_SD example

**Config**
Hypnos_SD example configk

prototypicalpro, 2019-11-12:
I was unable to replicate this issue, but I think I found the problem. Loom will only synchronize an RTC if it detects the RTC has been reset (i.e. the battery has been removed) or NTP is configured. Since you are not using NTP, I suspect this issue is a result of the RTC running for extended periods without synchronization because the battery was left in on accident. Can you remove the battery from the hypnos boards you tested and see if this resolves the issue?

maxchu765, 2019-11-12:
Yes, I've tried both removing the battery and re-inserting before uploading and uploading without the battery. Both situations still show the same issue. However, the date is correct in all the tests I've done.

maxchu765, 2019-11-12:
Could the RTC be synchronizing to cached time from an earlier compilation?

prototypicalpro, 2019-11-12:
Unlikely, unless the RTC is failing to communicate somehow. Can you leave the hypnos boards you tested in the Loom drawer (towards the entrance to the lab) so I can try and replicate the issue with them?

maxchu765, 2019-11-12:
I left two Hypnos boards in the Loom drawer. When testing, I get the correct time (offset by 2-3 minutes due to Issue #23) on the very first upload after opening the Arduino IDE. Even after removing the coin cell and unplugging the feather and Hypnos from power, any subsequent uploads have the wrong time,

prototypicalpro, 2019-11-18:
I found a typo in our timezone correction code (addressed in https://github.com/OPEnSLab-OSU/Loom/commit/a710619589cb3b1b89ed2e7875bfa763b0b0006a), however this should only contribute to the device being consistently an hour off after daylight savings, which doesn't sound like the issue you are having.

I think the symptoms are due to the RTC being synchronized by a Loom device with an old timestamp. When Loom initializes the DS3231 RTC, it performs the following steps in order:
1. Initialize the Wire library.
2. Print the time currently reported by the DS3231.
3. Read the DS3231 status register. If the oscillator stop flag is set (indicating "the oscillator either is stopped or was stopped for some period" according to the datasheet), set the RTC to the time Loom was compiled.
4. Read the RTC again. If the time is considered invalid (the year, month, or day are outside reasonable numbers, such as 2000 or 2050 for the year), set the RTC to the time Loom was compiled. 
5. Print the time.

If this process is working correctly, the RTC should synchronize once on the first time it is plugged in to *any* Loom device, and then never again so long as a battery is present. After some testing, I only found one instance where this process did not work correctly: the RTC synchronized again despite having already synchronized correctly, I suspect due to the oscillator stop flag being sensitive to some unknown electrical phenomenon. Other than that instance the process worked correctly.

With this in mind, there are two reasons I could come up with that the RTC would read an incorrect time:
 1. The oscillator stop flag is triggered due to some electrical noise during RTC operation, causing the RTC to synchronize again on a reset.
 2. The RTC is accidentally reset (or never synchronized) and then plugged into a Loom device which was uploaded to a long time ago.

Since you were experiencing this problem frequently, I suspect most of your issues are due to cause 2. Unfortunately, Loom as a software platform cannot detect or fix this cause, as Loom would have to know the RTC is incorrect by comparing it against another time source. 

Given this information, I can only suggest that you be cautious with when and how you reset your RTCs, and consider formalizing the process of synchronizing an RTC to happen once per day. If the problem persists please feel free to reopen this issue.

closed 2019-11-18 by prototypicalpro