OPEnSLab-OSU/Loom #49: Hypnos V2 Software Fix

nguyenb5, 2019-11-19:
**Describe the bug**
SPI pins leak current to 3VRAIL for Hypnos. 

**Hardware in Use**
Hypnos V2

**To Reproduce**
Init SD, write something to SD, shutdown 3Vrail, Loom Deepsleep

**Expected behavior**
3V LED still turn on dimly.

**Code**
See below

**Additional context**
Suggested to set all SPI pins to INPUT before going to sleep.

Here's an example sketch.
https://gist.github.com/kamocat/b4cf9f0ccbd765f78a6d7390d2da2fe8

Cogswatch, 2019-11-25:
The Hypnos page of the GitHub wiki has been updated to include the modified power up and power down procedure. Hypnos will continue to live adjacent to Loom rather than having a custom library implementation until further development can be committed.

nguyenb5, 2019-11-27:
Max has tested the board, and sd.begin needs to be run again after power-up by LOOM.

prototypicalpro, 2020-02-11:
An example has been created demonstrating this fix: https://github.com/OPEnSLab-OSU/Loom/blob/master/examples/Lab%20Examples/Hypnos_SD_Sleep/Hypnos_SD_Sleep.ino . It is still necessary to change the state of the state of the SPI pins in the sketch: we left it this way because Loom does not currently have a framework for ordering power up/down actions, and changing the state of the SPI pins too early on accident could result in a failure of another module using the SPI bus.

closed 2020-02-11 by prototypicalpro