OPEnSLab-OSU/Loom #64: SD Failed to Initialize

osuwalkera6, 2020-01-24:
**Describe the bug**
Code compiles and uploads successfully but fails to initialize the SD, thus no data is recorded. The sensors will still report data to the console, they just go unrecorded.

**Hardware in Use**
Arduino Feather M0
Hypnos

**To Reproduce**
Steps to reproduce the behavior:
1. Upload Hypnos_SD example to M0 board
2. Open Console

**Expected behavior**
The SD card should Initialize

**Code**
<pre>
<font color="#434f54">&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;</font>

<font color="#434f54">&#47;&#47; This is a basic example that demonstrates usage of the Hypnos board.</font>

<font color="#434f54">&#47;&#47; The Hypnos board includes</font>
<font color="#434f54">&#47;&#47; - SD</font>
<font color="#434f54">&#47;&#47; - DS3231 RTC</font>
<font color="#434f54">&#47;&#47; - Ability to power of peripherals</font>

<font color="#434f54">&#47;&#47; Further details about the Hypnos board can be found here:</font>
<font color="#434f54">&#47;&#47; </font><u><font color="#434f54">https:&#47;&#47;github.com&#47;OPEnSLab-OSU&#47;OPEnS-Lab-Home&#47;wiki&#47;Hypnos</font></u><font color="#434f54"></font>

<font color="#434f54">&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;&#47;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">Loom</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#434f54">&#47;&#47; Include configuration</font>
<font color="#00979c">const</font> <font color="#00979c">char</font><font color="#434f54">*</font> <font color="#000000">json_config</font> <font color="#434f54">=</font> 
<font color="#5e6d03">#include</font> <font color="#005c5f">&#34;config.h&#34;</font>
<font color="#000000">;</font>

<font color="#434f54">&#47;&#47; Set enabled modules</font>
<font color="#000000">LoomFactory</font><font color="#434f54">&lt;</font>
&#09;<font color="#000000">Enable</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Internet</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Disabled</font><font color="#434f54">,</font>
&#09;<font color="#000000">Enable</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Sensors</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Enabled</font><font color="#434f54">,</font>
&#09;<font color="#000000">Enable</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Radios</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Enabled</font><font color="#434f54">,</font>
&#09;<font color="#000000">Enable</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Actuators</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Enabled</font><font color="#434f54">,</font>
&#09;<font color="#000000">Enable</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Max</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">Enabled</font>
<font color="#434f54">&gt;</font> <font color="#000000">ModuleFactory</font><font color="#000000">{</font><font color="#000000">}</font><font color="#000000">;</font>

<font color="#000000">LoomManager</font> <font color="#000000">Loom</font><font color="#000000">{</font> <font color="#434f54">&amp;</font><font color="#000000">ModuleFactory</font> <font color="#000000">}</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#5e6d03">setup</font><font color="#000000">(</font><font color="#000000">)</font> 
<font color="#000000">{</font>
&#09;<font color="#434f54">&#47;&#47; Needs to be done for Hypno Board</font>
&#09;<font color="#d35400">pinMode</font><font color="#000000">(</font><font color="#000000">5</font><font color="#434f54">,</font> <font color="#00979c">OUTPUT</font><font color="#000000">)</font><font color="#000000">;</font>&#09;&#09;<font color="#434f54">&#47;&#47; Enable control of 3.3V rail </font>
&#09;<font color="#d35400">pinMode</font><font color="#000000">(</font><font color="#000000">6</font><font color="#434f54">,</font> <font color="#00979c">OUTPUT</font><font color="#000000">)</font><font color="#000000">;</font>&#09;&#09;<font color="#434f54">&#47;&#47; Enable control of 5V rail </font>

&#09;<font color="#434f54">&#47;&#47;See Above</font>
&#09;<font color="#d35400">digitalWrite</font><font color="#000000">(</font><font color="#000000">5</font><font color="#434f54">,</font> <font color="#00979c">LOW</font><font color="#000000">)</font><font color="#000000">;</font>&#09;<font color="#434f54">&#47;&#47; Enable 3.3V rail</font>
&#09;<font color="#d35400">digitalWrite</font><font color="#000000">(</font><font color="#000000">6</font><font color="#434f54">,</font> <font color="#00979c">HIGH</font><font color="#000000">)</font><font color="#000000">;</font>&#09;<font color="#434f54">&#47;&#47; Enable 5V rail</font>

&#09;<font color="#000000">Loom</font><font color="#434f54">.</font><font color="#000000">begin_serial</font><font color="#000000">(</font><font color="#00979c">true</font><font color="#000000">)</font><font color="#000000">;</font>
&#09;<font color="#000000">Loom</font><font color="#434f54">.</font><font color="#000000">parse_config</font><font color="#000000">(</font><font color="#000000">json_config</font><font color="#000000">)</font><font color="#000000">;</font>
&#09;<font color="#000000">Loom</font><font color="#434f54">.</font><font color="#000000">print_config</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>

&#09;<font color="#000000">LPrintln</font><font color="#000000">(</font><font color="#005c5f">&#34;\n ** Setup Complete ** &#34;</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> 
<font color="#000000">{</font>
&#09;<font color="#000000">Loom</font><font color="#434f54">.</font><font color="#000000">measure</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
&#09;<font color="#000000">Loom</font><font color="#434f54">.</font><font color="#000000">package</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
[... 16 lines cut ...]
&#09;&#09;<font color="#000000">&#39;name&#39;</font><font color="#434f54">:</font><font color="#000000">&#39;Device&#39;</font><font color="#434f54">,</font><font color="#000000">\</font>
&#09;&#09;<font color="#000000">&#39;instance&#39;</font><font color="#434f54">:</font><font color="#000000">1</font><font color="#434f54">,</font><font color="#000000">\</font>
&#09;&#09;<font color="#000000">&#39;interval&#39;</font><font color="#434f54">:</font><font color="#000000">5000</font><font color="#434f54">,</font><font color="#000000">\</font>
&#09;&#09;<font color="#000000">&#39;print_verbosity&#39;</font><font color="#434f54">:</font><font color="#000000">2\</font>
&#09;<font color="#000000">}</font><font color="#434f54">,</font><font color="#000000">\</font>
&#09;<font color="#000000">&#39;components&#39;</font><font color="#434f54">:</font><font color="#000000">[</font><font color="#000000">\</font>
&#09;&#09;<font color="#000000">{</font><font color="#000000">\</font>
&#09;&#09;&#09;<font color="#000000">&#39;name&#39;</font><font color="#434f54">:</font><font color="#000000">&#39;Analog&#39;</font><font color="#434f54">,</font><font color="#000000">\</font>
[... cut ...]
&#09;&#09;&#09;<font color="#000000">&#39;params&#39;</font><font color="#434f54">:</font><font color="#000000">&#39;default&#39;</font><font color="#000000">\</font>
&#09;&#09;<font color="#000000">}</font><font color="#000000">\</font>
&#09;<font color="#000000">]</font><font color="#000000">\</font>
<font color="#000000">}</font><font color="#000000">&#34;</font>

</pre>

**Additional context**
I have seen Max use the exact same board and Hypnos and it worked for him. I am unsure what is the cause .

prototypicalpro, 2020-01-28:
Changing the version of the Adafruit SAMD core (under `Tools->Board->Boards Manager->Adafruit SAMD Boards`) to **v1.5.7** fixes the issue. I will investigate what change caused this problem.

prototypicalpro, 2020-01-28:
I have filed an issue with the [Adafruit SAMD core](https://github.com/adafruit/ArduinoCore-samd/issues/209) and the [SdFat library](https://github.com/greiman/SdFat/issues/175) to address this issue.

diego76765, 2020-10-06:
Hello I have problem in Sd root error to make in display lcd how the solve it thank you

closed 2022-01-25 by WL-Richards