OPEnSLab-OSU/Loom #67: Loom does not compile with an error containing `invalid use of template-name 'SSLClient'`

prototypicalpro, 2020-01-30:
After recent changes, Loom may not compile any sketch, generating an error that looks something like:
```
...
C:\Users\User\Documents\Arduino\libraries\Loom\src/InternetPlats/InternetPlat.h:44:10: error: invalid use of template-name 'SSLClient' without an argument list

  virtual SSLClient& get_client() = 0;

          ^~~~~~~~~
...
```
This is a known issue due to incompatible versions of SSLClient. To fix this issue, you will need to upgrade your SSLClient library to a version above v1.3.0. To upgrade your SSLClient library:
1. Navigate to the Arduino Library Manager (`Sketch->Include Library->Manage Libraries...`) and wait for the window to load.
1. Enter `SSLClient` into the search box and press enter.
1. Click on the update button on the item labeled `SSLClient`.

If you continue to have issues after completing these instructions, please file another issue.

MrKangs, 2020-09-16:
@prototypicalpro  Is this bug still exists? If you, can you tell me where?

prototypicalpro, 2020-09-16:
This issue has been resolved with newer versions Loom and SSLClient. I left this issue open as a reminder for people to ensure they upgrade SSLClient while upgrading Loom, however there is probably no need for it anymore now that most of the lab has upgraded. I will go ahead and close it.

closed 2020-09-16 by prototypicalpro