OPEnSLab-OSU/Loom #74: Analog pin Interrupts not working in Loom

udellc, 2020-02-19:
**Describe the bug**
Vanilla Attach interrupt works, but not Loom for Analog pins

Talk with Liam

prototypicalpro, 2020-02-19:
I found the issue. [Loom_Interrupt_Manager::register_ISR](https://github.com/OPEnSLab-OSU/Loom/blob/fc36a8df101e18fceca88510ed7e9ceb4cd5ef64/src/Interrupt_Manager.cpp#L188) assigns the interrupt settings to the `int_settings` array using the pin number as an index. The `int_settings` array is of size 16. Unfortunately, the analog pins A2-A7 refer to the values between 16-45, causing a segfault when the register_ISR function is called with those pins. I need to do more research on this issue, but in the meantime ISRs on pins > A1 should be considered broken.

prototypicalpro, 2020-02-20:
I created a pull request to warn the user if their interrupt pin is out of bounds while we figure out how to solve this more permanently: https://github.com/OPEnSLab-OSU/Loom/pull/77.

akerr501, 2020-11-14:
This issue has been solved in version 2.5.0 which can be read about [here](https://github.com/OPEnSLab-OSU/Loom/releases/tag/2.5.0). There are still limitations to the combinations of pins that interrupts can be attached to, but these are systemic to vanilla interrupts. There is also a sketch that demonstrates how to use interrupts now.

closed 2020-11-14 by akerr501