OPEnSLab-OSU/Loom #78: Google Sheets Publish Different Tabs For Each LoRa Device

prototypicalpro, 2020-02-25:
**Is your feature request related to a problem? Please describe.**
The SitkaNet project is using Loom to collect data on several LoRa nodes, send this data to a LoRa hub, and then publish this data to Google Sheets. One aspect of this feature set is that the Google Sheet contains a tab for every device the hub is receiving from, limiting the confusion of data points in each sheet. 

The current Google Sheets publishing implementation determines which tab to use by looking at the device ID stored in the LoomManager instance:
https://github.com/OPEnSLab-OSU/Loom/blob/a797aab736a23f4d9dca8c98e5ebac76b4e8a356/src/PublishPlats/GoogleSheets.cpp#L77-L81
This device ID is not updated when the JSON is changed, and as a result does reflect a the device ID of a LoRa packet being received from a different device. This was intentional behavior, as it doesn't make sense to change the device name of the hub every time we receive a packet from the child—one consequence of this, however, is that a single device can never publish to a different tab, regardless of values in the JSON.

Some possible solutions to this problem include:
* Redefining how ComPlats put messages into Loom (perhaps into another array in the contents array?), and then creating tabs based on that structure.
* Allowing the device name to be changed in the JSON at any time, and realizing the consequences for this in other modules (Spool?).
* Changing specifically Google Sheets publish to pull the device name from the JSON, and pushing a general solution until later (not recommended).

I've implemented the third bullet for @maxchu765 in the [SitkaNetV2](https://github.com/OPEnSLab-OSU/Loom/tree/SitkaNetV2) branch as a temporary fix, but I believe we should consider implementing one of the first two bullets for the rest of the lab.

prototypicalpro, 2020-02-26:
Another proposed solution is to keep a nodeID on Loom Manager with a specified value that corresponds to the Hub. This would be the simplest solution, but as with bullet 1 would require addressing this new value in all other publishing platforms.

udellc, 2021-10-19:
Can we have a description of how this is now handled in Loom3?