OPEnSLab-OSU/Loom #79: Power-Cycle Ordering is not Specified

prototypicalpro, 2020-02-27:
Now that Loom is planning to break out the automatic initialization from `LoomManager::sleep()` into the sketch, a documentation page is needed to specify the ordering in which Loom functions should be called on wakeup. This includes things such as turning on the Hypnos Board, clearing the RTC interrupt, calling `Loom::power_up()` and so on. This page also needs to address modules that we do not have a formal ordering for, but may cause conflict (ex. Ethernet and LoRa).

BGoto808, 2020-03-13:
https://github.com/OPEnSLab-OSU/Loom/wiki/Loom-Power-Cycle-Function-Ordering

prototypicalpro, 2020-03-13:
Could you add some example code snippets that include interrupt operations and the RTC operations? Things like `reconnect_interrupt`

closed 2020-06-25 by BGoto808