OPEnSLab-OSU/Loom #95: SleepyDog is included but not used, causing compilation errors

prototypicalpro, 2020-04-07:
**Describe the bug**
The Adafruit SleepyDog library is included in Loom, but never actually used: 
https://github.com/OPEnSLab-OSU/Loom/blob/c5f7a950fa959fdf10c0a8a2875f87445dfd76d5/src/Sleep_Manager.cpp#L11-L15 https://github.com/OPEnSLab-OSU/Loom/blob/c5f7a950fa959fdf10c0a8a2875f87445dfd76d5/src/Sleep_Manager.cpp#L69-L77 I suspect SleepyDog was previously used but was removed during feature development.

As SleepyDog conflicts with FeatherFault (see https://github.com/OPEnSLab-OSU/FeatherFault/issues/5), it would be nice if this dependency could be removed to prevent future issues.

prototypicalpro, 2020-04-07:
On second pass, I did find a place where SleepyDog is used: https://github.com/OPEnSLab-OSU/Loom/blob/49d1a5dba4bc0ff070ccce1bcf7e7348dd8846d5/src/Manager.cpp#L361-L371 Do project in the lab use `Loom::nap`?

BGoto808, 2020-04-13:
Looking at the Github wikis for all of the projects, the only project I see that uses the Loom::nap function is Smartrock. However, when looking at their Github, they have a more recent code that they have for field testing that does not use the Loom::nap function. I would have to check in with them to see whether they are using the code that uses the Loom::nap function or the one without it. 

https://github.com/OPEnSLab-OSU/SmartRock/blob/master/SmartRock2.5/SmartRock2.5.ino

akerr501, 2020-07-15:
SleepyDog has been removed from Loom with version [2.4.0](https://github.com/OPEnSLab-OSU/Loom/releases/tag/2.4.0). The nap function in Loom Manager has been moved outside to an example [sketch](https://github.com/OPEnSLab-OSU/Loom/tree/master/examples/Lab%20Examples/Nap) that requires Adafruit SleepyDog to be downloaded outside of Loom, resolving any conflicts with FeatherFault

closed 2020-07-15 by akerr501