OPEnSLab-OSU/Loom #123: RTC doesn't account for daylight savings

BGoto808, 2020-09-08:
**Describe the bug**
The RTC conversion readings from local to UTC time does not account for daylight savings resulting in the time readings to be off by up to 1 hour depending on the time of year and time zone. 

**Hardware in Use**
Any hardware with an RTC (Hypnos, DS3231, PCF8523, etc.) and a Feather M0

**To Reproduce**
Steps to reproduce the behavior:
1. Use Hypnos_SD example code
2. Change config.h to use local time as PDT and use_utc_time to be true (10, true)
3. Upload code to Feather

**Expected behavior**
The time readings to accurately convert from local time to UTC time

**Code**
Look at [Hypnos_SD example code](https://github.com/OPEnSLab-OSU/Loom/tree/master/examples/Lab%20Examples/Hypnos_SD). 

**Config**
"{\
	'general':\
	{\
		'name':'Device',\
		'instance':1,\
		'interval':5000,\
		'print_verbosity':2\
	},\
	'components':[\
		{\
			'name':'Analog',\
			'params':'default'\
		},\
		{\
			'name':'SD',\
			'params': 'default'\
		},\
		{\
			'name':'DS3231',\
			'params':[10, true]\
		}\
	]\
}"

**Additional context**
Daylight savings time zones should have different conversions than the standard local time zone.

MrKangs, 2020-09-08:
There were some false readings in some time zone. I fixed that part so that it will be better. The user needs to update as daylight saving happens. This will be pushed into the code to master soon.

MrKangs, 2020-09-09:
We are planning to change for a checker rather than a user that needs to change in the config file. Reference can be found [here](https://www.instructables.com/id/Adding-Daylight-Savings-Time-to-Your-RTC/).
Also, we are planning that no matter user wants, UTC will be printed out rather than the local time zone. This will be updated soon.

MrKangs, 2020-09-15:
Now it has a checker feature that it will automatically switch standard time to daylight saving/ summer time when it happens on the day that it suppose to change. At the same time, we remove some time zone since they are not used anymore or incorrect names for that time zone. Last, for now on, the RTC will read UTC as default. Therefore, if you want to read your local time rather than UTC, then set the params as true in the second parameter. This code will be merged into master soon. If you need this feature/ bug fix, then please use the RTC branch for now.

closed 2020-09-15 by MrKangs