OPEnSLab-OSU/Loom #132: Loom Installation Error

osuwalkera6, 2021-02-07:
**Describe the bug**
Using the quick install code for Windows (curl -fsSL https://raw.githubusercontent.com/OPEnSLab-OSU/Loom/master/setup/setup-windows.sh | sh) an error message comes up:

"Updating loom:samd@2.4.0 with loom:samd@2.5.0...
Error during install: installing platform loom:samd@2.5.0: checking local archive integrity
main: line 34: open: command not found"

**Hardware in Use**
None

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Loom Quick Install webpage (https://github.com/OPEnSLab-OSU/Loom/wiki/Arduino-and-Loom-Quick-Setup)
2. Copy and paste code into GitBash
3. Press enter and wait for error to show up

**Expected behavior**
No errors to occur

**Code**
curl -fsSL https://raw.githubusercontent.com/OPEnSLab-OSU/Loom/master/setup/setup-windows.sh | sh

**Config**
none

**Additional context**
Add any other context about the problem here.

akerr501, 2021-02-07:
I can look into what might be causing this issue, but I would suspect it has to do something with updating. The board profile and install script are both relatively new and sometimes we miss potential bugs like this. In the meantime, if you need to update manually I would recommend going to the Tools->Boards->Board Manager and searching for Loom, and then update the version there.

udellc, 2021-02-07:
Thanks for looking into this! on Loom 2.5 we will have about 60 novice students trying to download Loom onto their computers for programming starting Spring term. If we can identify the issue and test before then, I think this will cause much fewer headaches for the the team in Spring.

closed 2022-01-25 by WL-Richards