OPEnSLab-OSU/Loom #139: STEMMA Inits, but reports only 65535

udellc, 2021-04-28:
**Describe the bug**
STEMMA inits okay. Temp value is good. But reports only 65535 for Capacitance value. 
Old STEMMA sensors work, but no new STEMMA sensors work

**Hardware in Use**
STEMMA, Adalogger M0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Sensor val should report 300 something as nominal value and go up to about 1000 when held in hand. 
**Code**
#include "Adafruit_seesaw.h"

Adafruit_seesaw ss;

void setup() {
  Serial.begin(115200);

  Serial.println("seesaw Soil Sensor example!");
  
  if (!ss.begin(0x36)) {
    Serial.println("ERROR! seesaw not found");
    while(1);
  } else {
    Serial.print("seesaw started! version: ");
    Serial.println(ss.getVersion(), HEX);
  }
}

void loop() {
  float tempC = ss.getTemp();
  uint16_t capread = ss.touchRead(0);

  Serial.print("Temperature: "); Serial.print(tempC); Serial.println("*C");
  Serial.print("Capacitive: "); Serial.println(capread);
  delay(100);
}

**Config**
No config here, just example code 

**Additional context**
It seems we need to update library and add new one
https://github.com/adafruit/Adafruit_BusIO (Links to an external site.)
https://github.com/adafruit/Adafruit_Seesaw (Links to an external site.)

Need to update Board profile to include these libraries and delete old Seesaw

BGoto808, 2021-09-26:
Seems that the library itself is the issue because testing the sensor with the Adafruit's example code produces the same problem (https://forums.adafruit.com/viewtopic.php?t=162556). Someone pointed out that there was an error with the touchread function and they modified it and got it working. 

'''
/*!
 ******************************************************************************
 *  @brief      read the analog value on an capacitive touch-enabled pin.
 *
 *  @param      pin the number of the pin to read.
 *
 *  @return     the analog value. This is an integer between 0 and 1023
 ****************************************************************************/
uint16_t Adafruit_seesaw::touchRead(uint8_t pin) {
  uint8_t buf[2];
  uint8_t p = pin;
  uint16_t ret = 65535;
  int x;

 /* do {
    Serial.println("at touchRead, before delay");
    //delay(1);
    //this->read(SEESAW_TOUCH_BASE, SEESAW_TOUCH_CHANNEL_OFFSET + p, buf, 2,1000);
    this->read(SEESAW_TOUCH_BASE, SEESAW_TOUCH_CHANNEL_OFFSET + p, buf, 2, 1000);
    ret = ((uint16_t)buf[0] << 8) | buf[1];
  } while (ret == 65535);
  */
/*modified 10-Apr-2021 by JimF to work on ESP32 DEVKIT V1 using Adafruit soilsensor.  Code worked with Arduino Mega 2560 but hung in read() when executed on ESP32. 
Issue seemed to be duration of optional delay between request sent to soilsensor and reading soilsensor.  if delay too brief, device returns 65535 and code hangs because of while loop.
Solution is to lengthen delay and only try a finite number of times. A delay of 5000 took only a single call to read to get
a valid result, where as a delay of 2500 took 2 reads, 1000 took 3 reads, and 500 took 4 or 8 reads. */
  for (x = 1; x <= 255; x++)
  {
      this->read(SEESAW_TOUCH_BASE, SEESAW_TOUCH_CHANNEL_OFFSET + p, buf, 2, 5000);
      ret = ((uint16_t)buf[0] << 8) | buf[1];
      if (ret != 65535) break;  //exit on valid result
  }
  // Serial.print("loop count ="); Serial.println(x);
  return ret;
}
'''

winniiew, 2021-10-05:
Commented out the touchRead() and replaced with

```
uint16_t Adafruit_seesaw::touchRead(uint8_t pin) {
uint8_t buf[2];
uint8_t counter=0;
uint8_t p = pin;
uint16_t ret = 65535;

do {
  delay(1);
  this->read(SEESAW_TOUCH_BASE, SEESAW_TOUCH_CHANNEL_OFFSET + p, buf, 2,
5000); //increasing the length of the buffer time
ret = ((uint16_t)buf[0] << 8) | buf[1];

counter++;
} while ((ret == 65535) & (counter<3));
return ret;
}
```

closed 2022-01-25 by WL-Richards