OPEnSLab-OSU/Loom #149: Test Loom_Neopixel - Winnie

winniiew, 2021-08-03:
Neopixel is able to flash one color but will not alternate even when using the same function

winniiew, 2021-08-03:
`///////////////////////////////////////////////////////////////////////////////

// This is simple example that is used to change the color of a 
// Neopixel to red, yellow, and green with a 2 second delay for each color

// This was wired through the A2 pin

///////////////////////////////////////////////////////////////////////////////

#include <Loom.h>

// Include configuration
const char* json_config = 
#include "config.h"
;

// In Tools menu, set:
// Internet  > Disabled
// Sensors   > Enabled
// Radios    > Disabled
// Actuators > Enabled
// Max       > Disabled
  
using namespace Loom;

Loom::Manager Feather{};

void setup() 
{ 

  pinMode(5, OUTPUT);
  digitalWrite(5, LOW); // Sets pin 5, the pin with the 3.3V rail, to output and enables the rail
  pinMode(6, OUTPUT);
  digitalWrite(6, HIGH); // Sets pin 6, the pin with the 5V rail, to output and enables the rail

  pinMode(A2, OUTPUT);

  Feather.begin_serial(true);
  Feather.parse_config(json_config);
  Feather.print_config();

  LPrintln("\n ** Setup Complete ** ");
}

void loop() 
{
  getNeopixel(Feather).set_color(2, 0, 200, 0, 0);
  delay(2000);
  getNeopixel(Feather).set_color(2, 0, 0, 0, 0);
  delay(2000);
  
  getNeopixel(Feather).set_color(2, 0, 200, 200, 0);
  delay(2000);
  getNeopixel(Feather).set_color(2, 0, 0, 0, 0);
  delay(2000);
  
  getNeopixel(Feather).set_color(2, 0, 0, 200, 0);
  delay(2000);
  getNeopixel(Feather).set_color(2, 0, 0, 0, 0);
  delay(2000);
  
  Feather.measure();
  Feather.package();
  Feather.display_data();
  Feather.pause();
}`

winniiew, 2021-08-03:
`"{\
  'general':\
  {\
    'name':'Device',\
    'instance':1,\
    'interval':2000\
  },\
  'components':[\
    {\
      'name':'NeoPixel',\
      'params':'default'\
    }\
  ]\
}"`

winniiew, 2021-08-03:
Fault! Cause: HARDFAULT
Fault during recording: No
Line: 117
File: Neopixel.cpp
Failures since upload: 13
Initialized Serial!

lukegoertzen, 2021-08-04:
Was a different config used for the case the caused the fault? It looks like a config that was only partially parsed, which resulted in no Neopixel being instantiated, trying to command it in that case causes a segfault.

winniiew, 2021-08-04:
> Was a different config used for the case the caused the fault? It looks like a config that was only partially parsed, which resulted in no Neopixel being instantiated, trying to command it in that case causes a segfault.

I used played around with different sketches with Bryson and that was the last sketch file I used which had the neopixel flashing green but no other color

winniiew, 2021-08-04:
Arduino: 1.8.15 (Windows 10), Board: "Loomified Feather M0, Arduino, Off, Wifi, Disabled, Enabled, Disabled, Disabled, On"

C:\Users\winni\OneDrive\Documents\Arduino\neopixeltest\neopixeltest.ino: In function 'void loop()':

neopixeltest:48:3: error: 'getNeopixel' was not declared in this scope

   48 |   getNeopixel(Feather).set_color(2, 0, 200, 0, 0);

      |   ^~~~~~~~~~~

exit status 1

'getNeopixel' was not declared in this scope

winniiew, 2021-08-04:
C:\Users\winni\OneDrive\Documents\Arduino\neopixeltest\neopixeltest.ino: In function 'void loop()':
neopixeltest:48:3: error: 'getNeopixel' was not declared in this scope
   48 |   getNeopixel(Feather).set_color(2, 0, 200, 0, 0);
      |   ^~~~~~~~~~~
exit status 1
'getNeopixel' was not declared in this scope

winniiew, 2021-08-06:
Initialized Serial!

[Device] deserializeJson() failed: InvalidInput
[Device] Config:
	Device Name         : Device
	Instance Number     : 1
	Device Type         : Node
	Interval            : 1000
[Device] Modules:

 ** Setup Complete **

lucasballr, 2021-08-10:
Seems like it was unable to finish setting up the modules. You might want to alter the config.h to see if anything fixes it.

winniiew, 2021-08-10:
Tested it w/o Hypnos board

Fault! Cause: HARDFAULT
Fault during recording: No
Line: 117
File: Neopixel.cpp
Failures since upload: 13
Initialized Serial!

[Device] deserializeJson() failed: InvalidInput
[Device] Config:
	Device Name         : Device
	Instance Number     : 1
	Device Type         : Node
	Interval            : 1000
[Device] Modules:

 ** Setup Complete **

lucasballr, 2021-08-10:
Could you post the config.h file?

winniiew, 2021-08-10:
Updated config.h

"{\
 'general':\
 {\
    'name':'Device',\
    'instance':1,\
    'interval':500\
  },\
  'components':[\
    {\
      'name':'Digital',\
      'params':'default'\
    },\
    {\
      'name':'Neopixel',\
      'params':'default'\
    }\
  ]\
}"

rcpeene, 2021-08-10:
There was a problem with the Config file; missing a comma.
BUT upon solving that, the issue returned to the original problem.
Featherfault identified a hard fault in Neopixel.cpp, Line 117.

When running the code, the first call to Neopixel::set_color() will run and complete. And then the sketch continues to run in loop(). However, any subsequent Neopixel method calls including set_color(), enable_pin(), print_state(), print_config() will not run and the code will seemingly halt before any of those functions are run.

Could this be that the Neopixel somehow gets uninitialized after its use? I don't see any other possible reasons the code would halt but only when Neopixel methods are called.

lukegoertzen, 2021-08-14:
I mentioned it in one of the meetings, but `getNeopixel(Feather)` (and similar functions) might warrant investigation. Those functions look a bit suspicious to me and are one of the changes introduced in v3, which might explain regular hanging being seen in v3. 
The implementation is:
`Loom::Neopixel getNeopixel(Loom::Manager feather) {return *(feather.get<Loom::Neopixel>());}` 
I would expect either a reference or pointer to be returned here (or ideally just use feather.get<X>() directly to avoid duplicate functions).

winniiew, 2021-08-17:
updated sketch
```
///////////////////////////////////////////////////////////////////////////////

// This is simple example that is used to change the color of a 
// Neopixel to red, yellow, and green with a 2 second delay for each color

// This was wired through the A2 pin

///////////////////////////////////////////////////////////////////////////////

#include <Loom.h>

// Include configuration
const char* json_config = 
#include "config.h"
;

// In Tools menu, set:
// Internet  > Disabled
// Sensors   > Enabled
// Radios    > Disabled
// Actuators > Enabled
// Max       > Disabled
  
using namespace Loom;

Loom::Manager Feather{};

void setup() 
{ 

  pinMode(5, OUTPUT);
  digitalWrite(5, LOW); // Sets pin 5, the pin with the 3.3V rail, to output and enables the rail
  pinMode(6, OUTPUT);
  digitalWrite(6, HIGH); // Sets pin 6, the pin with the 5V rail, to output and enables the rail

Feather.begin_serial(true);
Feather.parse_config(json_config);
Feather.print_config();

  LPrintln("\n ** Setup Complete ** ");
}

void loop() 
{

  //Neopixel->set_Color(2, 0, 200, 0, 0);     // Sets color to red
  Feather.get<Loom::Neopixel>()->set_color(2, 0, 200, 0, 0);
  delay(2000);
  
  //Neopixel->set_Color(2, 0, 200, 200, 0);   // Sets color to yellow
  Feather.get<Loom::Neopixel>()->set_color(2, 0, 200, 200, 0);
  delay(2000);
  
  //Neopixel->set_Color(2, 0, 0, 200, 0);     // Sets color to green
  Feather.get<Loom::Neopixel>()->set_color(2, 0, 0, 200, 0);
  delay(2000);
  
  Feather.measure();
  Feather.package();
  Feather.display_data();
  Feather.pause();
}
```

closed 2021-08-17 by rcpeene