OPEnSLab-OSU/Loom #157: Test Loom 3 with LoRa Communication Module - Winnie

winniiew, 2021-08-19:
[LoRa] Send failed

winniiew, 2021-08-19:
```
Initialized Serial!

= = = = = Parse Config = = = = =

Config Pretty Version:
{
  "general": {
    "name": "Device",
    "instance": 1,
    "interval": 2000
  },
  "components": [
    {
      "name": "Analog",
      "params": [
        8,
        12,
        true,
        true,
        true,
        true,
        true,
        true,
        0,
        0,
        0,
        0,
        0,
        0
      ]
    },
    {
      "name": "LoRa",
      "params": [
        255,
        0,
        23,
        3,
        200
      ]
    }
  ]
}
SIZE: 565
= = = = = Generate Objects = = = = =

[Device] Adding Module: Analog
[LoRa] 	Initializing Manager Success
[LoRa] 	Setting Frequency Success
[LoRa] 	Setting Power Level to 23
[LoRa] 	Setting retry timeout to 200 ms
[LoRa] 	Setting max retry count 3
[Device] Adding Module: LoRa
= = = = = = = = = = = = = = = = =

[Device] Config:
	Device Name         : Device
	Instance Number     : 1
	Device Type         : Node
	Interval            : 2000
[Device] Modules:
		[+] Analog
		[+] LoRa

 ** Setup Complete ** 
[Device] Json:
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.286133,
        "A[0]": 1809,
        "A[1]": 1627,
        "A[2]": 1615,
        "A[3]": 1470,
        "A[4]": 1656,
        "A[5]": 1661
      }
    }
  ]
}
[LoRa] Send failed
```

winniiew, 2021-08-20:
Commplat.h

Line 111
changed second parameter from 0 to 1000, to increase window time so packet would show up 

```
bool			receive(JsonObject json) { return receive_blocking(json, 1000); }
```

rcpeene, 2021-08-24:
A new problem has arisen for me.

The previous fix, changing 0 to 1000 works. However, including 'WiFi' as a component in the config with valid parameters in the LoRa sketch changes it. I have compiled the same basic example sketch with and without WiFi in the config and it stops working when WiFi is included in the config.

udellc, 2021-08-24:
Are we sure the JSON formatting of the config is done properly? Can you paste this in?

closed 2021-08-24 by rcpeene