OPEnSLab-OSU/Loom #160: Test Loom_TSL2561 - Winnie

winniiew, 2021-08-30:
TSL2561 failed to initialize

winniiew, 2021-08-30:
[Device] Adding Module: Analog
[TSL2561] Initialize failed
[Device] Cannot add inactive module

udellc, 2021-08-31:
The TSL 2561 is discontinued and no longer available to purchase anyways. We should retire this option in the code base.

lukegoertzen, 2021-08-31:
I would disagree about removing this sensor module (or other working code not causing any harm). Even if it can't be purchased directly anymore, any existing hardware should still work just fine. Additionally, as an open source project, there may (perhaps unlikely) be people using the module still. It doesn't hurt to leave the sensor support code in the code base, other than perhaps compiled code size, but based on the nRF + sensor compile issues seen, the granularity of module enable sets should probably be reevaluated - e.g. perhaps add a deprecated sensor factory option.

lucasballr, 2021-09-07:
We'll be removing this from the library, so no need to worry about this anymore.

closed 2021-09-08 by lucasballr