OPEnSLab-OSU/Loom #161: Test Loom_MAX31856 - Winnie

udellc, 2021-08-31:
Needs testing with K-type thermocouple wire

closed 2021-08-31 by winniiew

winniiew, 2021-09-07:
```
Cold Junction Temp: 25.98
Thermocouple Temp: 27.95
Cold Junction Temp: 26.03
Thermocouple Temp: 28.09
Cold Junction Temp: 25.98
Thermocouple Temp: 28.11
Cold Junction Temp: 26.02
Thermocouple Temp: 28.23
Cold Junction Temp: 26.03
Thermocouple Temp: 28.23
Cold Junction Temp: 26.09
Thermocouple Temp: 28.30
```

winniiew, 2021-09-07:
```
 ** Setup Complete ** 
[Device] Json:
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.305469,
        "A[0]": 1838,
        "A[1]": 1580,
        "A[2]": 1600,
        "A[3]": 1374,
        "A[4]": 1708,
        "A[5]": 1796
      }
    },
    {
      "module": "MAX31856",
      "data": {
        "temp": 26.1875
      }
    }
  ]
}
[Device] Json:
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 2
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.303857,
        "A[0]": 1809,
        "A[1]": 1618,
        "A[2]": 1638,
        "A[3]": 1361,
        "A[4]": 1695,
        "A[5]": 1755
      }
    },
    {
      "module": "MAX31856",
      "data": {
        "temp": 26.22656
      }
    }
  ]
}
```