OPEnSLab-OSU/Loom #164: Test Loom_LIS3DH - Winnie

winniiew, 2021-09-02:
Initializes but the values don't change

winniiew, 2021-09-02:
"contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 1
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.323194,
        "A[0]": 1975,
        "A[1]": 1776,
        "A[2]": 1707,
        "A[3]": 1539,
        "A[4]": 1726,
        "A[5]": 1792
      }
    },
    {
      "module": "LIS3DH",
      "data": {
        "ax": 25.49219,
        "ay": 25.49219,
        "az": 25.49219
      }
    }
  ]
}
[Device] Json:
{
  "type": "data",
  "id": {
    "name": "Device",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 2
      }
    },
    {
      "module": "Analog",
      "data": {
        "Vbat": 4.321582,
        "A[0]": 1920,
        "A[1]": 1751,
        "A[2]": 1724,
        "A[3]": 1542,
        "A[4]": 1717,
        "A[5]": 1794
      }
    },
    {
      "module": "LIS3DH",
      "data": {
        "ax": 25.49219,
        "ay": 25.49219,
        "az": 25.49219
      }

rcpeene, 2021-09-17:
The issue was some apparent confusion when implementing the LIS3DH with the Sparkfun dependency. From the dependency, the LIS3DH.begin() method returns 0 upon a success and greater values to signify an error. In Loom, the code treated begin() as type bool, and assumed that the Sparkfun dependency would return 0 on a failure (like most other sensors). So Loom was misinterpreting the dependencies return values and deactivating the Loom LIS3DH module.

It is worth noting that the Adafruit dependency (which Loom does not use as of right now) uses a traditional begin() scheme, returning 0 on a failure and 1 on a success. As far as I can tell, these dependencies are only compatible with their respective hardware. In other words, if Loom continues to use the Sparkfun dependency, it shall only work with the Sparkfun LIS3DH hardware, and vice versa with Adafruit's hardware and software.

closed 2021-09-17 by rcpeene