OPEnSLab-OSU/Loom #167: Test Loom_K30 - Winnie

winniiew, 2021-09-14:
Module [K30] does not have default settings, please provides params
[K30] Cannot add null module

winniiew, 2021-09-14:
config.h
```
"{\
  'general':{'name':'K30', 'instance':1,'interval':5000},\
  'components':[\
    {'name':'K30','params':'default'},\
    {'name':'SD','params':[true,1000,10,'K30',true]}\
  ]\
}"
```

winniiew, 2021-09-14:
Sketch
```
//////////////////////////////////////////////////////////////////////////////////////////////////////////

// This is basic example to use the K30 sensor with Feather.
// The K30 sensor will measure the CO2 from the range 0 to 10000 ppm and log to SD.

// In this example of program, it will measure the CO2 level and display on the Serial Monitor.
// Note that the K30 sensor needs at least 6 minutes of warm up to get accurate measurements.
// Therefore, you will get data values after 6 minutes have passed.
// Inside this program, there is a timer that will run for 6 minutes before it starts measuring values.

// As you might see, there are some syntax that came from outside of the Loom language,
// so take a look the link and the code carefully.
// https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/creating-a-new-serial

///////////////////////////////////////////////////////////////////////////////////////////////////////////

#include <Loom.h>
#include "wiring_private.h" // pinPeripheral() function

// Include configuration
const char* json_config =
#include "config.h"
;

// In Tools menu, set:
// Internet  > Disabled
// Sensors   > Enabled
// Radios    > Enabled
// Actuators > Disabled
// Max       > Disabled

using namespace Loom;

Loom::Manager Feather{};

// Create Serial SERCOM for K30 Sensor: RX pin 12, TX pin 11
Uart Serial2 = Uart(&sercom1, 12, 11, SERCOM_RX_PAD_3, UART_TX_PAD_0);

void setup() {

  while (!Serial);
  pinMode(5, OUTPUT);
  digitalWrite(5, LOW); // Sets pin 5, the pin with the 3.3V rail, to output and enables the rail
  pinMode(6, OUTPUT);
  digitalWrite(6, HIGH); // Sets pin 6, the pin with the 5V rail, to output and enables the rail

  Serial2.begin(9600);

  Feather.begin_serial(true,true);
  Feather.parse_config(json_config);
  Feather.print_config();

  //Assign pins 10 & 11 SERCOM functionality
  //pinPeripheral(10, PIO_SERCOM);
  //pinPeripheral(11, PIO_SERCOM);

  //Feather.K30().set_serial(&Serial2);
  Feather.get<Loom::K30>()->set_serial(&Serial2);
  

  LPrintln("\n ** Setup Complete ** ");

  warmUpTimer();

}

void loop() {

  Feather.measure(); // Sample attached sensors
  Feather.package(); // Format data for display and SD
  Feather.display_data(); // display printed JSON formatted data on serial monitor
  Feather.get<Loom::SD>()->log(); // Loggin K30 Data value into SDCard
  Feather.pause();
}

void SERCOM1_Handler(){ // This function is require for the K30 Serial Sensor because of UART Type
  Serial2.IrqHandler();
}

void warmUpTimer(){ // This function is a timer to warm up the K30 sensor to get accurate measurements

  LPrintln("\n ** Set up 6 minutes Warm Up time to get accurate measurements ** ");

  LPrintln("\n Hung ");

  for(int timePassed = 1; timePassed < 7; timePassed++){ // By pausing Loom, it will not measure CO2 value for 6 minutes
    Feather.pause(60000); // The max is only 1 min for pause, we loop it for 6 times to make it 6 minutes
    LPrint(timePassed); // Knowing the User that how many minutes have been passed
    LPrint(" minute(s) passed!");
    LPrint("\n");
  }

  LPrintln("\n ** Ready to Measure ** ");
}

rcpeene, 2021-09-14:
First issue "Cannot add null module" has been fixed by changing a line in Loom's K30.h.
Changing the registration function call at the bottom of the file from REGISTER_NODEFAULT() to REGISTER() fixes that issue. The example being worked with above attempts to add the K30 module with 'default' parameters, though this cannot be done when the module is register with REGISTER_NODEFAULT().

udellc, 2021-10-18:
@SmithJar-rov , please test by Week 5 meeting - if enough time with K30 hardware arriving.

winniiew, 2021-10-29:
Issue: pins were not correctly connected on the K30
```
[K30] Json:
{
  "type": "data",
  "id": {
    "name": "K30",
    "instance": 1
  },
  "contents": [
    {
      "module": "Packet",
      "data": {
        "Number": 3
      }
    },
    {
      "module": "K30",
      "data": {
        "CO2": 552
      }
    }
  ]
}
```

closed 2021-10-29 by winniiew