OPEnSLab-OSU/Loom #168: Test Loom_MB1232 - Winnie

BGoto808, 2021-09-24:
According to page 2 of the datasheet (https://www.maxbotix.com/documents/I2CXL-MaxSonar-EZ_Datasheet.pdf), there is one pin that needs to be pulled HIGH or LOW depending on the task its doing which is pin 2 on the sensor. Looking at the Loom code, there is no way for the user to specify which pin that is being used for that and as a result, the program hangs when the Feather tries to communicate with the sensor.

udellc, 2021-11-08:
Let's make this low priority. No need to hold up Loom 3 on this.

winniiew, 2021-11-17:
Note: Try to find an external library that supports I2C

udellc, 2021-11-17:
I think we can ignore the Pin 2 thing and can use the Loom i2c functionality as-is. It's possible the default address of the sonar was changed during testing. Looking at the example code at the bottom of the datasheet, there are functions to set a new default address in addition to functions for taking a sonar reading. I think someone in CS with i2c experience could translate the bare example code into loom formatted functions. We would need to change the address back to factory default ourselves.

udellc, 2021-11-17:
I am inclined to table this until after the Loom 3 release though.

closed 2022-01-25 by WL-Richards