OPEnSLab-OSU/Loom #181: Add ability to read serial number from board to be used as unique identifier

kurtjd, 2022-03-14:
**Is your feature request related to a problem? Please describe.**
A new feature planned for the SmartRock project is to have SD log files given a unique identifier to match it with a physical SmartRock device in case files/SD cards get jumbled. A possible solution I discovered for this was to read the unique serial number every SAMD21 chip (the chip on Feather M0s) is given at specific memory addresses and use this as an identifier. This functionality could potentially be useful for many projects so it might be worth adding it to Loom.

**Describe the solution you'd like**
I have written a function that reads the 128-bit identifier from memory and converts it to a 32 character hexadecimal string to match the same serial number given in Arduino IDE (Tools->Board Info). This makes it easy to match filenames to board identifiers. It would need a few modifications to fit into the Loom library, but it currently looks like:

```cpp
char serial_no[33];

void readSerialNumber() {
  uint32_t sn_words[4];
  sn_words[0] = *(volatile uint32_t *)(0x0080A00C);
  sn_words[1] = *(volatile uint32_t *)(0x0080A040);
  sn_words[2] = *(volatile uint32_t *)(0x0080A044);
  sn_words[3] = *(volatile uint32_t *)(0x0080A048);

  for (int i = 0; i < 4; i++) {
    for (int j = 0; j < 4; j++) {
      sprintf(serial_no + (i * 8) + (j * 2), "%02X", (uint8_t)(sn_words[i] >> ((3 - j) * 8)));
    }
  }
}
```

**Describe alternatives you've considered**
Originally we had considered using EEPROM to store hand-created serial numbers for each board, and then that value could be read to be used as the unique filename for data files. However, Feather M0s don't actually have EEPROM, and the best you can do is use Flash storage to emulate EEPROM. Unfortunately, Flash memory is erased every time a sketch is uploaded (there may be ways around this by writing startup scripts that reserve part of Flash memory so it is not overwritten) but that is a bit of a hassle when it seems just reading the unique serial number already there is much simpler.

**Additional context**
I don't know how often this functionality was ever needed in other projects using Loom so it might not be worth adding it just to have it. However, it is pretty simple to implement so it could prove useful in the future. When a Loom manager instance is created, it could call that readSerialNumber function to store the serial number in memory, then it could be accessed by something like ```Feather.get_serial_no()```

udellc, 2022-03-17:
My concerns with a 32 char serial number is too long to be used as a file name, and eye-crossing for associating what that number is with a specific device.
We could pull the serial number and save it onto its own file on the SD card called serialnum.txt that logs the serial number and the date in which it was operational. Then the user could cross reference this with the dates on their data log files.
Even better perhaps,
WeatherChimes already automatically  manages how data is labeled using the device name and number. This could be automatically done for SD data log files. instead of needing to put a specific file name in the SD config parameter or in the .ino

kurtjd, 2022-03-29:
> My concerns with a 32 char serial number is too long to be used as a file name, and eye-crossing for associating what that number is with a specific device. We could pull the serial number and save [...]

@colinhalebrown 
@udellc 
Ok I will get with Colin and let him know of this alternative. However, besides file names, do you see any potential uses for being able to access board serial numbers easily via Loom? If not, I will go ahead and close this issue, otherwise I could add it to the library pretty easily.

udellc, 2022-03-29:
If you could add this functionality to Loom as a "grabber" function that the user could grab and then use in their ino code or log to SD manually, this would be fine for now. It would give Colin the ability to do what he wants and make an open door for using this info in the future.

closed 2022-04-22 by kurtjd