OPEnSLab-OSU/Loom #194: Loom doesn't set time until after sample period

BGoto808, 2022-05-11:
**Describe the bug**
When manually setting the RTC time through the serial monitor, the time doesn't get set for the clock until after the sample period which may cause issues if the user doesn't wait until the end of the sample period. 

**Hardware in Use**
Feather + Hypnos (or any RTC device)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Hypnos_SD_Sleep example
2. Compile and upload code to Feather
3. Boot up serial monitor and enter in time values as it prompts
4. Disconnect Feather before the program puts it to sleep

**Expected behavior**
The time will not be set and will reprompt the user for it the next time

kurtjd, 2022-05-12:
Is this using the Loom 2 example or the Loom 3 one? This is surprising since from what I remember when I was digging around in the Loom RTC code is that the time gets set before setup() ends and loop() even starts, but who knows, might be something weird going on.

udellc, 2022-05-12:
Yes I would have thought the user-input time from serial is prompted in Setup - so it *should* set the time before leaving Setup. But according to Bryson's usage it does not. Do you think you can reproduce the issue for Kurt, Bryson?

BGoto808, 2022-05-12:
This is for Loom 3. You can grab a dendrometer node from the lab and reproduce it by uploading the [Dendrometer node code](https://github.com/OPEnSLab-OSU/Dendrometer/tree/V3_Testing/AS5311_Prototype/Prototype_V1.1/node) on the GitHub. The code is setup so that the magnet needs to be in a good position before moving onto the loop. 

So you can test it by setting the time and the code will automatically stop when the sensor tries to read the magnet position.

udellc, 2022-05-12:
Maybe we can look at this sometime next week?

BGoto808, 2022-05-12:
I can send an updated testing guide for using the Dendrometer node to test this issue so that someone can test it next week.

kurtjd, 2022-05-12:
I will see if I can reproduce this without Dendrometer first. If it happens with just a simple Feather + Hypnos setup it's likely a Loom issue. I'll update this with what I find.

kurtjd, 2022-05-16:
I was not able to replicate this using a Feather M0 WiFi and Hypnos v3.3. Using the Loom 3 Hypnos_SD_Sleep example, I first reset the RTC and as expected I was prompted to enter a custom time. I immediately removed power from the Feather and after plugging it back in, it seemed the time was set correctly and it did not reprompt me.

Thinking that maybe the Feather went to sleep before I could unplug it, I reset the RTC again and I put in a 10 second delay right before the Feather would go to sleep to ensure I had time to remove power before sleep. Again, after trying this, the RTC seemed to be set correctly despite losing power before going to sleep.

So this doesn't appear to be a general Loom issue. Is there any other information you have that might further help narrow this down?